SaddlebackCollegeRobotics/.github #10: Object detection algorithm

Supernova1114, 2023-10-16:
[URC 2024 Rules](https://drive.google.com/file/d/1Y_lltxyXzpqOE1q90gFQ-IBEvniJOQxL/view)
In this year's autonomy mission, there are requirements for object detection other than Aruco tags.
We will need to detect an orange mallet, and a plastic reusable water bottle.
You can find specific in the rules including a link to the mallet item.

We will need to find or create a method to detect these items.
In addition to the detection algorithm, we will want the script to be wrapped into a ROS 2 service node.
- First check to see if there are already any libraries that support ROS 2

- [ ] Find / Create object detection algorithm
- [ ] Initially test it with common objects.
- [ ] ROS support? / Write ROS service.

ROS 2 service:
- Input = image
- Output = ???

Supernova1114, 2023-11-02:
Update: 
- Nvidia has object detection packages in the Isaac ROS library. Support for Yolov8 and detectnet model.
- Support for ROS
- Hardware accelerated
- https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_object_detection

Supernova1114, 2023-11-10:
ToDo:
- [ ] Order orange mallet
- [ ] Order extra water bottles / colors for training.

Supernova1114, 2023-11-13:
Doing research:
https://www.kaggle.com/code/abdallahwagih/bottle-synthetic-classification-acc-99
https://www.youtube.com/watch?v=m9fH9OWn8YM
https://www.kaggle.com/code/abdallahwagih/water-bottle-efficientnetb3-85-71
https://storage.googleapis.com/openimages/web/index.html

kavehzare98, 2024-01-09:
Progress Report:
- We have a working model that can identify most objects (specifically bottles) based on camera feed.
- Current issue: pre-trained model cannot identify hammers or the orange mallet.

Things To Do:

- [ ]  Gather our own images on Saturday, January 13th, 2024 for the orange mallet (10,000 images) and water bottles (5,000 images).
- [ ] Annotate our dataset via Roboflow by January 16th.
- [ ] Train model on custom dataset on Google Colab
- [ ] Test model on orange mallet and Nalgene bottles in appropriate environment using ZED
- [ ] Configure Object's distance from rover via bounding boxes and frame coordinates
- [ ] ROSify and test on rover
- [ ] (Optional) Configure obstacle avoidance

Supernova1114, 2024-03-22:
Labeling is in progress.

Supernova1114, 2024-05-25:
Object detection implementation is done, including a distance estimator.

closed 2024-05-25 by Supernova1114