SaddlebackCollegeRobotics/.github #97: Research MoveIt (Experimental)

Supernova1114, 2023-10-16:
Used for arm automation and control.

https://moveit.ros.org/

Supernova1114, 2023-11-21:
Did some research, very cool tech.
Will probably have to be a summer research project.

Supernova1114, 2023-12-28:
Update:
Cam Wolff did some research.
Looks like MoveIt will be the easiest way to get IK working for the arm.

Terracom12, 2023-12-29:
@Supernova1114 
*Unsure how much this has already been looked into, so I'll just lay out what I've found.*

MoveIt has a guide specifically tailored to the type of movement it seems we're after. The image below shows the controller configuration in the out-of-the-box example. 
[image]
The guide for this setup can be found on this [MoveIt documentation page](https://moveit.picknik.ai/main/doc/how_to_guides/controller_teleoperation/controller_teleoperation.html#).

To get this working:
- [ ] Configure controller
  - *8BitDuo has strange configurations for the left and right triggers in that their outputs range \[-1, 1\], which should probably be mapped \[0, 1\]*
- [ ] Setup arm geometry and constraints. 
  - [Guide here](https://moveit.picknik.ai/main/doc/examples/examples.html#integration-with-a-new-robot). Haven't looked much into this yet.
- [ ] Docker integration (?)
  - Here's the [official guide](https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html) on setting this up. I managed to get this working very easily on my Debian machine. 
  - **NOTE: This Docker configuration is very bulky, meant for working on a system with a GUI \(rviz\). However, it should be fairly simple to create a custom lightweight image.**

Supernova1114, 2024-01-12:
Research move-it studio

kavehzare98, 2024-01-22:
> YIPPE

nice