SaddlebackCollegeRobotics/.github #135: Web GUI Design

Supernova1114, 2023-11-10:


devloos, 2023-11-15:
- [Web Socket Tutorial](https://www.youtube.com/watch?v=cw8-KFVXpTE)
- [Django Python Folder Structure](https://medium.com/django-unleashed/django-project-structure-a-comprehensive-guide-4b2ddbf2b6b8)

devloos, 2024-01-28:
- Video feed of the rover: from the zed
- Watchdog status: is he starving or not
- Command console
- GPS map
- Live Lidar scans
- Antenna signal strength
- Different feature setups/layouts for different missions

devloos, 2024-01-28:
General:

-  Battery level
- Drive speed
- Current
- Longitude and latitude 
- Heading 
- Distance traveled
- Military time clock
- Antenna signal strength

Arm:

- Visual model of the current position of the arm and its axis
- Retract the arm

Science:

- Sensor data
- Sensor visualization, i.e graphs

Autonomy:

- Bounding box for detected object
- Input for GPS coordinates
- Map height and waypoint visualization
- Rover position and ability to add points to map by clicking
- Box with a colour that is the same as the one on the rover that flashes when the rover arrives at a destination, drives autonomously, etc...
- Green pathfinding line from ROS navigation stack

devloos, 2024-01-28:
- Video feed of the rover
  - Feed from zed
  - Arm cameras
- Watchdog status
  - We will probably need 10
  - Should we have the ability to reset?
- Command console
  - is there a library
  - lets restrict some commands
- GPS map
  - something like boat GPS system
  - [image]
- Live Lidar scans
  - can I integrate this with gps
  - I should search for a library
- Different feature setups/layouts for different missions

devloos, 2024-02-06:
Terminal libraries offering complete SSH capabilities cannot be seamlessly integrated into another project. In simpler terms, such libraries function as standalone projects and do not integrate well with our existing projects.

While it is possible to implement a basic shell for executing and transmitting commands, there are limitations. This approach only supports basic commands without the ability to edit or navigate through the folder structure.

devloos, 2024-02-07:
Foxglove:

It seems that the most effective way to establish a real-time connection is through the utilization of web sockets. The system supports diverse layouts, including the incorporation of tabs within each layout.

Key features for our needs:
Video feeds
Map display
Lidar visualization
Various metrics visualization options, such as watchdog status, antenna signal strength, drive speed, and more (not limited to heading)
Service calls
Custom message schemas

Limitations to consider:
Inability to perform action calls
Incapacity to read or set ROS parameters

The implementation process is relatively straightforward but necessitates an additional node, namely 'foxglove_bridge', to be executed. This particular node actively listens to topics being published by other nodes. While certain node settings can be adjusted to modify behavior, ultimately, the data is transmitted through a web socket.

Despite these intricacies, setting up a video feed and visualizing the robot took only a few minutes to accomplish.

Also the video feed you are seeing is compressed and sent via a standard ros2 sensor msg "CompressedImage"

Foxglove understands most if not all std msgs out the gate

closed 2024-04-12 by Supernova1114