SaddlebackCollegeRobotics/.github #136: Web GUI Feature Design

Supernova1114, 2023-11-10:


Supernova1114, 2023-11-12:
Some examples: 
- Video feed
- Command console

Sam-Nasri, 2023-12-01:
General:

-  Battery level
- Drive speed
- Current
- Longitude and latitude 
- Heading 
- Distance traveled
- Military time clock

Arm:

- Visual model of the current position of the arm and its axis
- Retract the arm

Science:

- Sensor data
- Sensor visualization, i.e graphs

Autonomy:

- Bounding box for detected object
- Input for GPS coordinates
- Map height and waypoint visualization
- Rover position and ability to add points to map by clicking
- Box with a colour that is the same as the one on the rover that flashes when the rover arrives at a destination, drives autonomously, etc...
- Green pathfinding line from ROS navigation stack

closed 2023-12-03 by Supernova1114

devloos, 2024-01-08:
- Video feed of the rover
  - Feed from zed
  - Arm cameras
- Watchdog status
  - We will probably need 10
  - Should we have the ability to reset?
- Command console
  - is there a library
  - lets restrict some commands
- GPS map
  - something like boat GPS system
  - [image]
- Live Lidar scans
  - can I integrate this with gps
  - I should search for a library
- Antenna signal strength
- Different feature setups/layouts for different missions