SaddlebackCollegeRobotics/.github #189: Fix odrive over-current issues

Supernova1114, 2024-04-12:


Supernova1114, 2024-04-24:
https://discourse.odriverobotics.com/t/odrive-s1-current-limit-violation-error/11432

Made a post on the ODrive forums regarding the issue.
Need to respond.

Supernova1114, 2024-04-26:
Tweaked the PID loop for the new motor gearbox.
Only saw around 2 Amp current spikes while free spinning. (Down from 20 A spikes).

Supernova1114, 2024-04-26:
Potentially setting the vel_integrator value to zero may be a fix for our issues.

Supernova1114, 2024-05-25:
Going to put this on hold, we implemented a function to reset the drive and arm system if needed.
It looks like we would need to have integrated encoders with better resolutions.

void-aere, 2025-10-23:
Developer mode was enabled on drive, and `config.inverter0.current_hard_max` set to `10000`, which seems to have eliminated overcurrent errors.

There are current protections on the DC bus, and the ODrives are configured not to draw over 40 amps continuous, which should make this decision a safe one.