SaddlebackCollegeRobotics/.github #208: Implement hi-res image pub for science

Supernova1114, 2024-05-24:


Terracom12, 2024-10-17:
Capture an image to fulfill the science mission's requirements for site analysis:

Requirements for this are stated as follows:
> - A wide-angle panorama showing the full context of the site. The panorama must
indicate cardinal directions, and have some indication of scale.
> - A close up, well focused, high-resolution picture with some indication of scale
(scale can be indicated post-capture) at the sampling site.
> - Teams will be required to take a stratigraphic profile using the on-board cameras
to determine the depositional environment and history of water in the sedimentary
structures visible at the site.

Start by just implementing the panoramic shot, then cardinal direction can easily be added based on GPS readings obtained at that time point.

nate0m, 2025-01-22:
See [Pull Request](https://github.com/SaddlebackCollegeRobotics/science/pull/2)